brainflow-dev/brainflow #553: Brainflow with ThinkPulse sensors on OpenBCI helmets

sonoshrmsi, 2022-09-20:
I am using an OpenBCI helmet and decided to change the sensors from the original 16 sensors to ThinkPulse sensors, and the code that I was using in Python is not working like before. The error says that the streaming is not happening, and whenever I want to run the code and save the data, the dataset is empty which was not the case with the original sensors. However, in the OpenBCI GUI, I am getting signals and data which means the hardware setup is A-okay. 

I will appreciate it if you could help me to run my code and get data again with Python.

Andrey1994, 2022-09-22:
openbci gui uses brainflow, so it should work if you call it directly also. Share stdout and logger output and all other relevant info and I will take a look. also make sure that gui is closed

closed 2022-09-22 by Andrey1994

julienconscious, 2022-12-12:
Hi, please check that you were able to change the gain from default X24 to X8 otherwise signal might be railed and you will end up with no signal. Best.

sonoshrmsi, 2022-12-14:
@julienconscious Thank you very much for your response. We are getting the data now, but the issue is that we are not able to change the settings in Python. How can we change the default gains in Python?

Andrey1994, 2022-12-14:
There is config board method, for exact command check openbci docs or logs in openbci gui