brainflow-dev/brainflow #554: update simpleble

Andrey1994, 2022-09-25:


rahulbiswas876, 2022-09-26:
@Andrey1994 can you provide the description of the issue? Willing to take

Andrey1994, 2022-09-26:
Great thanks!

BrainFlow uses SimpleBLE(https://github.com/brainflow-dev/brainflow/tree/master/third_party/SimpleBLE) for muse devices, to build it with simpleble need to provide --ble argument to tools\build.py
Recently SimpleBLE was updated significantly. Before There were dependencies like SimpleBLUEZ SimpleDBUS etc. Now they moved to single project and also fixed some bugs.

The challenge is that I made some custom changes in SimpleBLE used in brainflow, e.g. https://github.com/brainflow-dev/brainflow/blob/master/third_party/SimpleBLE/CMakeLists.txt#L233 https://github.com/brainflow-dev/brainflow/blob/master/third_party/SimpleBLE/include/simpleble_c/shared_export.h etc

PR #574 (2022-11-03): add support for ganglion via native ble
closes #554 
closes #568
closed 2022-11-03 by Andrey1994