brainflow-dev/brainflow #555: Move synthetic data streams that would have lower sampling rates to SyntheticAux

philippitts, 2022-10-01:
**Use case:**
Emulating devices with synthetic or algorithmic data is useful in the development of code that consumes Brainflow. Many boards group data streams with slower sampling rates into the auxiliary and/or ancillary board variations. Currently data streams that often have slow sampling rates are in the default Synthetic Brainflow board definition, but it would make more sense for them to be in the auxiliary variation. Doing this would allow synthetic devices to better mirror real ones, and subsequently would allow client code to better abstract synthetic and real devices.

**Describe the solution you'd like:**
Move the battery, eda, ppg, accel, gyro, resistance and temperature channels to the SyntheticAux board definition.

Andrey1994, 2022-10-01:
makes sense but moving may break playback of synthetic board for already recorded files and some existing user code which currently gets it from default preset.
probably instead moving it can be copied to auxiliary preset with smaller sampling rate

Andrey1994, 2022-10-01:
and its ok if some data types are duplicated between presets, especially for synthetic board

closed 2022-10-04 by Andrey1994