brainflow-dev/brainflow #558: Cannot restart board streams when recording file is specified.

philippitts, 2022-10-07:
**Describe the bug**
Board streams cannot be restarted when a file is provided for recording.

**To Reproduce**
The following modification to the "brainflow_get_data" C++ example demonstrates the bug:

```
board->prepare_session ();
board->start_stream (450000, "file://test.csv:w");
//board->start_stream ();

Sleep (5000);

board->stop_stream ();
BrainFlowArray<double, 2> data = board->get_current_board_data (10);
std::cout << data << std::endl;

board->start_stream (450000, "file://test.csv:w");
//board->start_stream ();

Sleep (5000);

board->stop_stream ();
data = board->get_current_board_data (10);
board->release_session ();
std::cout << data << std::endl;
```

In this case the following error is logged:
```
[board_logger] [error] only one streamer per preset is currently supported
```

Note that the following cases will not fail:

1. `start_stream()` is called without a file parameter both times.
2. `start_stream()` is called with a file parameter the first time, but not the second time.

**Info (please complete the following information):**
 - Board Id: -1 (synthetic board)
 - OS: Windows 10
 - Programming Language and its version: C++ 17.1.3
 - Architecture(x86, x64, ARM, etc): x64
 - Brainflow Version: master (https://github.com/brainflow-dev/brainflow/commit/7f32586274ad009dcc8bc979268a63f1784a4c0d)
 - Brainflow Version: 5.3.0 (occurs on both master and release tag).

**Expected behavior**
The stream should be restarted even when a recording file is specified and the recording should function in the same way regardless of how many times the stream is restarted.

Andrey1994, 2022-10-07:
Actually, streamer params inside `start_stream` method is legacy and I was going to remove it but kept for backward  compatibility

There is a method `add_streamer` which should be used instead

I will think what to do with start_stream in the meantime you can just use add_streamer method

Andrey1994, 2022-10-07:
The questions are:

* should it be removed from start_stream
* when should we destroy streamers, before it was done in the beginning of start_stream method. Now with `add_streamer` method which can be called before its not a good option. It can be done in `stop_stream` or new method can be added like `delete_streamer`.

And #516 should be merged also

closed 2022-10-21 by Andrey1994