brainflow-dev/brainflow #559: Crown Brainflow unable to start stream

GabeManzo, 2022-10-10:
I have been attempting to set up Brainflow on the crown through my laptop but keep getting BOARD_NOT_READY_ERROR: 7 unable to start streaming session.

I am on a Mac on Monterey 12.6. I have python 3.7.13 and my crown is working perfectly. Furthermore I normally work on my university wifi but I have managed to connect it privately to my computer through a hotspot, would that suffice?

code used:
import brainflow

from brainflow import BoardIds
from brainflow.board_shim import BoardShim, BrainFlowInputParams
from brainflow.data_filter import DataFilter, FilterTypes, AggOperations

def main ():
    board_id = BoardIds.CROWN_BOARD.value # or BoardIds.NOTION_2_BOARD.value or BoardIds.NOTION_1_BOARD.value
    params = BrainFlowInputParams ()
    params.board_id = board_id
    BoardShim.enable_dev_board_logger ()
    board = BoardShim (board_id, params)
    board.prepare_session ()
    board.start_stream ()
    data = board.get_board_data ()
    board.stop_stream ()
    board.release_session ()

    print (data)

if __name__ == "__main__":
    main ()

Output:
(eeg-notebooks) manzo@wc-dhcp57d219 eeg-notebooks % python3 brainflow-test.py
[2022-10-10 14:06:41.005] [board_logger] [info] incoming json: {
    "board_id": 23,
    "file": "",
    "ip_address": "",
    "ip_port": 0,
    "ip_protocol": 0,
    "mac_address": "",
    "master_board": -100,
    "other_info": "",
    "preset": 0,
    "serial_number": "",
    "serial_port": "",
    "timeout": 0
}
[2022-10-10 14:06:41.006] [board_logger] [trace] Board object created 23
[2022-10-10 14:06:41.007] [board_logger] [debug] Use IP port 9000
[2022-10-10 14:06:46.007] [board_logger] [error] no data received in 5sec, stopping thread
[2022-10-10 14:06:46.007] [board_logger] [error] errno 35 message Resource temporarily unavailable
Traceback (most recent call last):
  File "brainflow-test.py", line 22, in <module>
    main ()
  File "brainflow-test.py", line 14, in main
    board.start_stream ()
  File "/Users/manzo/Downloads/ENTER/envs/eeg-notebooks/lib/python3.7/site-packages/brainflow/board_shim.py", line 1122, in start_stream
    raise BrainFlowError('unable to start streaming session', res)
brainflow.exit_codes.BrainFlowError: BOARD_NOT_READY_ERROR:7 unable to start streaming session

Andrey1994, 2022-10-10:
check this https://github.com/brainflow-dev/brainflow/issues/525 
@alexcastillo

Andrey1994, 2022-10-10:
actually it looks like a duplicate, closing it here, feel free to ask questions if any in 525

closed 2022-10-10 by Andrey1994

Andrey1994, 2022-10-10:
> I normally work on my university wifi but I have managed to connect it privately to my computer through a hotspot

to be honest I dont know and cannot test such setup. Maybe Alex can. More likely its the issue and you cannot send broadcasts in uni net