brainflow-dev/brainflow #578: Migrate to .NET Standard

javidsho, 2022-11-18:
**Use case:**
.NET Framework is deprecated and almost nobody uses it for new projects. Since most people either use .NET Core/.NET, this will result in package incompatibility and warnings when included in a project.

**Describe the solution you'd like:**
I'd really like cross-platform packages based on .NET Standard 2.0 - which is the most compatible of all, and that includes Unity

**Describe alternatives you've considered:**
I'm currently referencing the package and getting the warning. I'm still able to use brainflow, but I cannot use some packages due to incompatibility with .NET Standard such as logging!

Andrey1994, 2022-11-19:
Hi!

Could you send a PR with required changes and I will take a look and test? As far as I understand it should be a small fix

javidsho, 2022-11-27:
Hi there! Sure I'll do it when I have time

Andrey1994, 2022-12-09:
for others as a reference: https://code-maze.com/differences-between-net-framework-net-core-and-net-standard/

Andrey1994, 2022-12-09:
https://www.michaeltaylorp3.net/porting-a-net-framework-library-to-net-standard/

closed 2022-12-12 by Andrey1994