brainflow-dev/brainflow #588: Q: how to check if BrainBit works properly and turned on ? I tested device using Java package.

blandger, 2023-01-03:
Hi.
I cloned repo to my Linux, I've built C++ with instructions, when I built java exaamples by Maven. When I had to tweak builing JAR (brainflow-jar-with-dependencies.jar) to run java examples properly (.SO libs are put into wrong path inside JAR for linux).

Anyway I was able to run built example by command line like 

`java -cp ./target/brainflow-jar-with-dependencies.jar brainflow.examples.BrainFlowGetData --master-board 7`

I was amaized by results in logs. I've got two log attached.
First I run example when BrainBit device is trurned ON, then I run BB device when it's turned OFF completely (batery was unpluged).
I amaized that two logs are not empty and there is no error getting data for the second case.

Below is a updated code example (BrainFlowGetData) with more data logging.
```
public static void main (String[] args) throws Exception
    {
        BoardShim.enable_board_logger ();
        BrainFlowInputParams params = new BrainFlowInputParams ();
        System.out.println("Input Params = " + args.length);
        int board_id = parse_args (args, params);
        System.out.println("Parsed boardId = " + board_id);
        BoardShim board_shim = new BoardShim (board_id, params);

        board_shim.prepare_session ();
        // board_shim.start_stream (); // use this for default options
        board_shim.start_stream (450000, "file://file_stream.csv:w");
        BoardShim.log_message (LogLevels.LEVEL_INFO.get_code (), "Start sleeping in the main thread");
        Thread.sleep (5000);
        board_shim.stop_stream ();
        System.out.println("brdDataCount = " +  board_shim.get_board_data_count ());
        double[][] data = board_shim.get_current_board_data (30); // doesnt flush it from ring buffer
        // double[][] data = board_shim.get_board_data (); // get all data and flush
        // from ring buffer
        for (int i = 0; i < data.length; i++)
        {
            System.out.println("i = " + i);
            System.out.println(Arrays.toString (data[i]));
        }
        board_shim.release_session ();
    }
```

Attached is a log from console when device is OFF.
[brainBit_turned_off_log.txt](https://github.com/brainflow-dev/brainflow/files/10338335/brainBit_turned_off_log.txt)
Local file_stream.csv is also full of data.

What is the possible reason for such strange behavior ? Is library generates fake data ?
Thank you.

Andrey1994, 2023-01-05:
In your logs: Parsed boardId = -1
-1 is a synthetic board which generates data, you didnt provide board id properly, master board parameter is used only for streaming and playback boards

Andrey1994, 2023-01-05:
here is a part from example:

```
            if (args[i].equals ("--board-id"))
            {
                board_id = Integer.parseInt (args[i + 1]);
            }
```

closed 2023-01-05 by Andrey1994

blandger, 2023-01-06:
I tried parameter board-id, it gives error **UNSUPPORTED_BOARD_ERROR**. It's not supported by linux ?? _BrainBit doesnt support Linux._

file: libsimpleble-c.so is not found in jar file
file: libBrainFlowBluetooth.so is not found in jar file
Input Params = 2
Parsed boardId = 7
[2023-01-06 15:19:04.800] [board_logger] [info] incoming json: {"ip_address":"","ip_address_aux":"","ip_address_anc":"","mac_address":"","serial_port":"","ip_port":0,"ip_port_aux":0,"ip_port_anc":0,"ip_protocol":0,"other_info":"","timeout":0,"serial_number":"","file":"","file_aux":"","file_anc":"","master_board":-100}
[2023-01-06 15:19:04.800] [board_logger] [error] BrainBit doesnt support Linux.
Exception in thread "main" brainflow.BrainFlowError: Error in prepare_session:UNSUPPORTED_BOARD_ERROR
	at brainflow.BoardShim.prepare_session(BoardShim.java:1361)
	at brainflow.examples.BrainFlowGetData.main(BrainFlowGetData.java:21)

Andrey1994, 2023-01-06:
As log message and error code say, libneurosdk which is used by brainflow for brainbit does not support linux. 

There is brainbit bled board which does not use libneurosdk and works on linux too but it requires bled111 dongle. 

Check supported boards page at brainflow docs