brainflow-dev/brainflow #595: WSACleanup is getting called more times than WSAStartup for Crown

jaweedbrainiac, 2023-01-16:
**Describe the bug**
I'm using the .NET binding of brainflow with Neurasity Crown. When the BOARD_NOT_READY_ERROR exception is thrown 3 times, then I won't be able to use my HttpClient anymore. In simpler words, the internal WSAStartup counter is 4 (in my case) and it seems to me like after trying 3 times and getting error, each time probably WSAStartup is called once and WSACleanUp is called twice, which exhausts the internal counter and WSA resources are cleaned up.

**To Reproduce**
To reproduce in C++ or any other low-level lang:
1. Do a WSAStartup, Now if you do a socket operation it succeeds.
2. Try to connect to a non-existing CROWN_BOARD and wait until you get a BOARD_NOT_READY error. Make sure to release_session afterwards.
3. Now the socket operation won't work anymore

**Info (please complete the following information):**
 - CROWN_BOARD
 - Windows
 - C# .NET 6
 - x64

**Expected behavior**
WSAStartup and WSACleanup should be called exactly the same number of times.

**Additional context**
I checked the C++ source of brainflow, and it seems to me that all the trouble is happening in `broadcast_client.cpp`.
There's a potential source of bug there. Imagine calling the `close()` method of BroadCastClient multiple times. It calls WSACleanup every single time! I think there should be a private BOOL field indicating whether or not WSAStartup has been called

Andrey1994, 2023-01-16:
Hi!

You are right, and its not only on broadcastclient, in other socket helpers its the same. And you proposed solution is correct, need to set flaf is wsastartup was called successfully and reset this flag in close method.

Do you want to send a PR with this fix?

jaweedbrainiac, 2023-01-20:
Hi Andrey!

Thank you for your response. I was wondering if you could fix it since I don't have as much experience with c++ and since you mentioned the same problem is there in other places which I don't know, I believe it would be much better if you could find the time do it.

Best regards

PR #598 (2023-01-22): fix wsacleanup called more times than it should

closes #595 
this bug potentially exists for all boards with socket connection, and can be a problem if there are other libraries for tcp\ip inside your app
closed 2023-01-22 by Andrey1994