brainflow-dev/brainflow #604: Windows build error (MLModule.dll copy step)

kcoul, 2023-01-28:
**Describe the bug**
Build of BrainFlow target, in Release mode, fails with:

Error copying file (if different) from "C:/Users/username/repos/brainflow/compiled/Release/MLModule.dll" to "C:/Users/username/repos/brainflow/python_package/brainflow/lib/MLModule.dll".

Upon inspecting the source folder, MLModule.dll is located at "C:/Users/username/repos/brainflow/compiled/MLModule.dll", not in a "Release" subfolder. This explains the problem.

**To Reproduce**
Build in CLion for Windows.

Andrey1994, 2023-01-30:
Hi, we do not support clion oficially... and as far as I understand clion uses mingw build tools to compile projects on windows by default. And mingw also is not currently supported(#255).
More likely it will start working with clion as soon as #255 is fixed but I cannot provide the ETA for it.

If you do not like visual studio I think you can download VS community edition and use it only for building code and continue using clion for development - https://www.jetbrains.com/help/clion/quick-tutorial-on-configuring-clion-on-windows.html#MSVC

closed 2023-01-30 by Andrey1994

kcoul, 2023-01-30:
Good to know! For the record, CLion can in fact use Visual Studio for a generator (it uses Ninja generator by default). By using both the Visual Studio toolchain *and* the Visual Studio generator, the post-build copy step is successful.