brainflow-dev/brainflow #619: add method to rereference channels

Andrey1994, 2023-03-26:
https://pressrelease.brainproducts.com/referencing/

Andrey1994, 2023-03-29:
Hi and thanks for willing to implement it!

I think method signature should look like this: `void reference(BrainFlowArray<double, 2> data, std::vector<int> channels_to_reference, std::vector<int> reference_channels)`

That should be done for all supported languages. 

In terms of implementation first of all you need to average data from reference_channels and after that subtract the averaged data from channels_to_reference

In BrainFlow in general we implement everything in C\C++ and call methods from dyn libs from other languages, but here since implementation is very simple(just average multiple arrays and subtract) and since it requires 2d arrays I think it's ok to just implement it for each language

Andrey1994, 2023-05-16:
Hi and thanks for working on it!

It should be a static method of DataFilter class for all languages

kkashiva, 2023-05-27:
Hi @Andrey1994 The reference method implementation in DataFilter class is ready for review for the java_package.
I'll work on similar logic to implement this in other languages (Java is my strongest, so I started there).

I just have 1 doubt on where will this method be called from?

sorenjmadsen, 2023-12-19:
Anyone done this one in Python? Happy to implement in the dyn lib too