brainflow-dev/brainflow #620: emotibit: SCR channels

devriesewouter89, 2023-04-07:
**Use case:**
When using the Brainflow SDK (I use python), I'm able to get the EDA channel from the emotibit, yet no SCR: frequency, amplitude & rise channels. Using the [Emotibit Oscilloscope](https://github.com/EmotiBit/ofxEmotiBit) I'm able to get these channels.

**Describe the solution you'd like:**
Would it be possible to expand the Ancillary Preset of the Emotibit board with these channels?

Andrey1994, 2023-04-12:
Yes, its possible.
Could you explain what scr is and how to use it?

devriesewouter89, 2023-04-12:
Hi
I've misinterpreted the usage of scr. Apparently scr is a derivative of EDA, not vice versa: https://www.reddit.com/r/EmotiBit/comments/twcap7/how_do_i_interpret_the_skin_conductance_response/

The scr metrics are useful for insights in stimuli response FYI, yet since it's a derivative, I'm not sure if it should be in brainflow sdk...

Andrey1994, 2023-04-13:
as far as I understand its calculated from raw eda values. in this case it should not be a part of BoardShim class.
We can add method to calculate it from eda signal into DataFilter class if you share reference implementation or detailed algorithm description