brainflow-dev/brainflow #623: Example Unity code from Doc gives errors on Mac Unity 2022.2.15

99bits, 2023-04-28:
I'm testing the example SimpleGetData.cs that is in the documentation on a Mac with Unity 2022.2.15 and got these errors:

OverflowException: Arithmetic operation resulted in an overflow.
SimpleGetData.Update () (at Assets/SimpleGetData.cs:47)and many of these:
BrainFlowError: GENERAL_ERROR:17
brainflow.BoardShim.get_board_data_count (System.Int32 preset) (at <6a05e3999b6843f892860b7887c12381>:0)
brainflow.BoardShim.get_current_board_data (System.Int32 num_samples, System.Int32 preset) (at <6a05e3999b6843f892860b7887c12381>:0)
SimpleGetData.Update () (at Assets/SimpleGetData.cs:45)

99bits, 2023-04-29:
Changed this line in the update loop:
Debug.Log("Num elements: " + data.GetLength(1));
to:
Debug.Log("Num elements: " + data.GetLength(0));

and now getting: 
Num elements: 32
UnityEngine.Debug:Log (object)
SimpleGetData:Update ()

Andrey1994, 2023-05-01:
I cannot reproduce(but I have no mac and run with synth board) and I do not think that dimension printed in debug logs can affect it

closed 2023-05-01 by Andrey1994