brainflow-dev/brainflow #628: Support for Mac M1 and M2 processors, ARM64

bermarkus, 2023-05-15:
I use Brainflow on windows and linux but also need to run my applications on the latest Mac laptops. At the moment I cant run my applications on latest Mac.
A clear and concise description of what the problem is and why do you need this feature

**Describe the solution you'd like:**
I would need a Brainflow compiled release for MAC ARM64

**Describe alternatives you've considered:**
You are my only hope

Andrey1994, 2023-05-15:
Hi! 

Binaries released by brainflow for Mac are universal binaries and should work on x86 and arm Macs. Macos versions supported are 10.15+

Which programming language do you use and how?

bermarkus, 2023-05-15:
Oh thanks for clarifying.

We have a project using Brainflow, Timeflux, MNE, and making a portable folder using Pyinstaller. 

Will close the issue for now, none of us have Macs, just trying to create compatible version for our users.

bermarkus, 2023-06-22:
[image]

We've been trying to get freeEEG32 running on Mac M2. Each time we get error, see screenshots. When pulling the USB plug while running the script we find that it gets stuck on line 1144.  We were able to run synthetic board, but not freeEEG32. 

Can you help? The same example script runs with FreeEEG32 on windows, linux, MacAMD64, but not MacARM64. Does it mean that the library only doesn't work for FreeEEG32 or also not for other boards on MacARM64? We do not have other boards to test.

Mac Mini M2
Ventura 13.2.1 
Anaconda latest version, python 3.11
Latest install of brainflow

the script we used:
[brainflowtest.py.txt](https://github.com/brainflow-dev/brainflow/files/11839731/brainflowtest.py.txt)

Andrey1994, 2023-06-22:
https://brainflow.readthedocs.io/en/stable/SupportedBoards.html#cyton 

Read note about tty ports, I think here its the same

Andrey1994, 2023-06-22:
I can add similar comment for freeeeg to brainflow docs

closed 2023-06-22 by Andrey1994

bermarkus, 2023-06-22:
It works now, thanks so much!! 
I think it will help others if you add it to the docs.