brainflow-dev/brainflow #642: add activity index calculated from accelerometer data

Andrey1994, 2023-06-18:
reference implementation - https://github.com/obss/BIOBSS

Andrey1994, 2025-05-26:
Hi, you can try, I will write up how it may look like later today

MigCast9, 2025-05-31:
Hi, wondering if you had the chance to get a few more details on this. No issues if not, but thought I'd double check

Andrey1994, 2025-05-31:
Hi, sorry forgot about it on vacation.

So, the way it should look like:

1. add helpers methods to calculate some derivative metrics from the raw accel data - there are some ideas here https://github.com/obss/BIOBSS/tree/main/biobss/imutools that should be added to DataHandler on the low level interface and to all bindings to DataFilter, will be similar to method like `get_custom_band_powers` for eeg but will operate on IMU data. There will be multiple methods for different IMU metrics.
2. Add a method to create full IMU feature vector(for features which gonna be used for calculations), in fact it will call methods from step 1 and combine results in a single array. That also should be added into all high level languages
3. There are two options to calculate activity index: analytical approach with some equations on top of feature vector(check code in BIOBSS), I dont know the best way to handle it, you will need to research a little bit. For classifier it will look like this:
* Add enum to indicate to which limb IMU sensor is attached(Head, arm, finger, etc)
* Record data and publish it somewhere
* Start with smth like log regression, it will convert binary classifier into probability(score) and its what we want, more complicated stuff can be done on top.
4. Both options for point 3 should be implmented as a part of MLModel: you will need to check that https://github.com/brainflow-dev/brainflow/tree/master/src/ml 

Good luck and feel free to ask questions if any!

Andrey1994, 2025-05-31:
Also, feel free to create a first PR as soon as you have smth for point 1, its already will be valuable even wo next steps

closed 2026-09-13 by Andrey1994