brainflow-dev/brainflow #643: Add support for EGIAmpServerPro

shivam-199, 2023-06-29:
We have a 64-channel EGIAmpServerPro device. We would appreciate it if there is support for that device as well. 

If not, how would we be able to connect the device ourselves?

Andrey1994, 2023-06-29:
Hi, I will be glad to add this board, but I have no such device and will not be able to test the code. 
So, you will need to do it by yourself(I will help) or at least help me with testing and validation

https://brainflow.org/2022-11-01-adding-new-boards/ here is the link describing the process

shivam-199, 2023-06-29:
Hi @Andrey1994, I'll be happy to help you test it on the device. Please let me know what to do!

Andrey1994, 2023-06-29:
I will need full info about device such as communication protocol, existing API, etc

Smth like this for reference https://docs.openbci.com/Cyton/CytonDataFormat/

shivam-199, 2023-06-29:
There aren't many guides for this. But I have been using [LSL](https://github.com/labstreaminglayer/App-EGIAmpServer) to get data up to 1000 Hz. It does not have a provision for anything more than 1000. [BCI2000](https://www.bci2000.org/mediawiki/index.php/Contributions:AmpServerProADC) has provided another guide. I am also attaching a command guide for you to look over.
[ASProSDK_v30_NetworkAPIs_rman_8200031-00_20190716.pdf](https://github.com/brainflow-dev/brainflow/files/11903351/ASProSDK_v30_NetworkAPIs_rman_8200031-00_20190716.pdf)

Andrey1994, 2023-07-09:
I am looking at these samples and docs and it seems like it very difficult to do it without access to device or sdk, data format is not simple as well as commands.

julama, 2025-11-07:
Hi Andrey, I wanted to ask the current state of this task? It would be great to have EGI integrated in brainflow.
I have access to an EGI amp and can help with testing and/or coding. What ist the best way to go forward?