brainflow-dev/brainflow #648: Ioctls Header Location

stellarpower, 2023-07-20:
[In os_serial_ioctl.cpp](https://github.com/brainflow-dev/brainflow/blob/f8b42716f775271e0b8d1c464153778284a6b50d/src/utils/os_serial_ioctl.cpp#L14) on linux we look for the ioctls header under `/usr/include/asm/ioctls.h`.

However, on [Ubuntu](https://packages.ubuntu.com/search?suite=jammy&arch=any&searchon=contents&keywords=ioctls.h) and [Debian](https://packages.debian.org/search?searchon=contents&keywords=ioctls.h&mode=exactfilename&suite=bookworm&arch=any) this file is located in `/usr/include/asm-generic/ioctls.h`.

The result of this is that `NO_IOCTL_HEADERS` gets set, and Brainflow will return an error for various operations on the board. I have built the docker image as included in the repository, and inserted a static assertion into the build that showed that `NO_IOCTL_HEADERS` was set, so can confirm this is reproducible in official builds under linux.

For me, the application I am using that links against brainflow is unable to set the baud rate on the serial port for my board, so refuses to use it, and this fatally prevents me from using the application. So I think it is probably important to cover all bases when it comes to valid potential locations cross-platform.

Is the path `/usr/include/asm/ioctls.h` standard? Or should the macro be expanded to include other locations in these cases?

Andrey1994, 2023-07-23:
Hi!

I will fix it, could you also specify the board with such problem?

Andrey1994, 2023-07-23:
actually it should not affect precompiled binaries which we distribute, only if you build it from the source code

Andrey1994, 2023-07-23:
@stellarpower could you try a fix from #650 ? I have no board nor ubuntu to give it a try

stellarpower, 2023-07-24:
Sure! I will try as soon as I can, but, that just looking at the source code I think that will definitely work.

Will that work okay with other Linux distros or architectures? I was thinking it may be possible to test for both paths.

Andrey1994, 2023-07-24:
In theory it should since I added it at the bottom and under defined(NO_IOCTL_HEADERS) but I didnt test it

stellarpower, 2023-07-24:
Oh yes, I see what you mean now that I am on desktop.

Looks good to me. I haven't tested it against a board yet but it builds and the macro is set properly so I see no reason why not.

All I think that's missing is a comment above each section explaining the reason for the duplication. I would add it myself, but trying to open a pull request on a branch that's already in a pull request and I give up with git :sweat_smile: 

Here's some text if you want to use it:
```
// linux, also need to check that system headers installed
// Look first in /usr/include/asm for platforms where the ioctls header is installed here (see below)
#elif defined(__linux__) && !defined(__ANDROID__)
...
```

```
// If not found in /usr/include/asm, try looking in /usr/include/asm-generic
// If found above, NO_IOCTL_HEADERS is not set, so we skip this step.
// Othrwise, we search again here under the new path
#if defined(__linux__) && !defined(__ANDROID__) && defined(NO_IOCTL_HEADERS)
#if defined __has_include
#if __has_include(<sys/ioctl.h>) && __has_include(</usr/include/asm-generic/ioctls.h>) && __has_include(</usr/include/asm-generic/termbits.h>)
#include <sys/ioctl.h>
...

```

```

stellarpower, 2023-07-24:
Oh yes, I see what you mean now that I am on desktop.

Looks good to me. I haven't tested it against a board yet but it builds and the macro is set properly so I see no reason why not.

All I think that's missing is a comment above each section explaining the reason for the duplication. I would add it myself, but trying to open a pull request on a branch that's already in a pull request and I give up with git :sweat_smile: 

Here's some text if you want to use it:
```
// linux, also need to check that system headers installed
// Look first in /usr/include/asm for platforms where the ioctls header is installed here (see below)
#elif defined(__linux__) && !defined(__ANDROID__)
...
```

```
// If not found in /usr/include/asm, try looking in /usr/include/asm-generic
// If found above, NO_IOCTL_HEADERS is not set, so we skip this step.
// Othrwise, we search again here under the new path
#if defined(__linux__) && !defined(__ANDROID__) && defined(NO_IOCTL_HEADERS)
#if defined __has_include
#if __has_include(<sys/ioctl.h>) && __has_include(</usr/include/asm-generic/ioctls.h>) && __has_include(</usr/include/asm-generic/termbits.h>)
#include <sys/ioctl.h>
...

```

closed 2023-07-25 by Andrey1994