brainflow-dev/brainflow #649: OpenVibe support

bermarkus, 2023-07-21:
Have you considered adding support for OpenVibe? It would be wonderful if OpenVibe could acquire through Brainflow.

Cheers,
Bernard

Andrey1994, 2023-07-23:
Hi Bernard!

I thought about it before, and I am agreed that it would be great. But at this point I think it should be OpenVIBE engineers who implement it, at least unless there is a paid request to do it.

First of all, we already provide everything whats needed for the integration(C++ SDK, docs, and code samples)
And as far as I know OpenVIBE project is founded by the university and has much more all kinds of resources than we have.
Finally, its up to them to merge a PR and maintain this code in the future.

If there are volunters willing to implement it I will be glad to help and answer all questions regarding brainflow part of things. Our slack workspace is a good place to start such conversations.

closed 2023-07-23 by Andrey1994

bermarkus, 2023-07-23:
Hi Andrey,

I agree that it should come from OpenVibe, and I have messaged them first, but you replied first :)

I'll see if we can move forward if they reply and are willing to merge a PR, maybe I can motivate a volunteer with some modest funds.

Cheers,
Bernard