brainflow-dev/brainflow #652: CMake Library Exports

stellarpower, 2023-07-28:
**Describe the bug**
Unfortunately, there are a million ways to do anything in CMake. But, in modern CMake, usually it's preferred to export libraries from a package using the `PackageName::SpecificLibrary` syntax.

E.g.:
```CMake
find_package(MathFunctions 3.4 COMPONENTS Addition SquareRoot)

add_executable(myexe main.cc)

target_link_libraries(myexe PRIVATE MathFunctions::Addition MathFunctions::SquareRoot)
```
Derived from [CMake documentation](https://cmake.org/cmake/help/latest/guide/importing-exporting/index.html#exporting-targets)

In Brainflow, it's recommended to import them by expanding a variable that the CMake install script has prepared:
```CMake
target_link_libraries (
    brainflow_get_data PUBLIC
    # for some systems(ubuntu for example) order matters
    ${BrainflowPath}
    ${MLModulePath}
    ${DataHandlerPath}
    ${BoardControllerPath}
)
```
[(Source)]( https://github.com/brainflow-dev/brainflow/blob/5.8.2/cpp_package/examples/get_data/CMakeLists.txt )

Is there a particular reason for this design? Is it preferrable over something closer to the expected standard, like:
```CMake
target_link_libraries (
    brainflow_get_data PUBLIC

    brainflow::ml
    brainflow::data_handler
    brainflow::board_controller
)
```
?
As I think this would facilitate integrating with other tools where this is the expected use. I am also not a CMake expert but I think using just one module (e.g. the ML module), it should be able to work out any dependencies (the core library)

Thanks!

**Info (please complete the following information):**
 - Ubutnu 22.04
 - C++
 - AMD64

stellarpower, 2023-07-28:
Also, this should be a separate issue, but whilst on topic as maybe there is a motivating reason - the headers seem to be installed to `/usr/local/inc` for me - not under `/usr/local/include` as is standard (`inc` is not a standard path under /usr or /usr/local, etc. on linux). 

I would think `/usr/local/include/brainflow` would be the best location, and then I can `#include` them with `<brainflow/board_controller.h>`. Including with e.g. `"board_controller.h"` then means these aren't sitting in a directory just for Brainflow and are amongst other headers on the system

Andrey1994, 2023-07-28:
installation dir is controled by -DCMAKE_INSTALL_PREFIX parameter in cmake, and in build.py there is a way to provide it

stellarpower, 2023-07-28:
Then we should probably change [these lines]( https://github.com/brainflow-dev/brainflow/blob/5.8.2/CMakeLists.txt#L16 ) from 
```CMake
SET (PACKAGE_LIB_INSTALL_DIR ${CMAKE_INSTALL_PREFIX}/lib)
SET (PACKAGE_INC_INSTALL_DIR ${CMAKE_INSTALL_PREFIX}/inc)
```
 to 
```Cmake
SET (PACKAGE_LIB_INSTALL_DIR ${CMAKE_INSTALL_PREFIX}/lib/brainflow)
SET (PACKAGE_INC_INSTALL_DIR ${CMAKE_INSTALL_PREFIX}/include/brainflow)
```
At least on linux. I am not sure if this way things are now is more suitable for MacOS or Windows, so it may need to make a choice based on the platform. Alternatively, CMake may provide some way to detect the most suitable directory for headers or shared/static libraries given an install prefix.

I think placing libraries in the root of `${CMAKE_INSTALL_PREFIX}/lib` is fine - but as the filenames don't include the word "Brainflow" then it probably makes more sense to have a separate directory. I.e. if I find a file in /usr/lib called `libDataHandler.so`, then it would be hard to work out what it belongs to.

Andrey1994, 2023-07-28:
you can add "brainflow" word to your CMAKE_INSTALL_PREFIX variable. I see no issue using inc instead include

stellarpower, 2023-07-28:
That wouldn't work, because then it'd go to /usr/local/brainflow/inc and /usr/local/brainflow/lib. The standard locations for a linux system would be /usr/include/brainflow  and /usr/lib/brainflow if installed by the package manager and typically /usr/local/include/brainflow and /usr/local/lib/brainflow for a manual installation. /usr/inc and /usr/local/inc do not exist.

closed 2023-10-01 by Andrey1994