brainflow-dev/brainflow #654: Overload Filters for Outputting in-place/using new Buffer

stellarpower, 2023-07-28:
It seems from an API perspective that some of the filters in `data_handler.cpp` modify the data in-place, so that the output from the filter is written over the `data` input buffer. For my use case, I'd like them to write to a separate output buffer - and I think that being able to specify this would be useful in general. Other filters aren't following this behaviour.

Some are using one technique or another to create an output buffer, then writing the values back over the input buffer:
- wavelet denoising allocates a new buffer and then [copies the values back one-by-one]( https://github.com/brainflow-dev/brainflow/blob/5.8.2/src/data_handler/data_handler.cpp#L492 )

Meanwhile, 
- The rolling filter() [adds samples one-by-one and then overwrites the input buffer.]( https://github.com/brainflow-dev/brainflow/blob/5.8.2/src/data_handler/data_handler.cpp#L425  )
- Downsampling() overwrites the buffer as it goes along.
- get_csp outputs two values, so takes two output buffers as arguments anyway.
- Similarly for the FFT and IFFT

So these could be modified easily to write output to a new buffer

The IIR/FIR filters seem to overwrite in the underlying implementation. It looks like the [interface for DSPFilters]( https://github.com/vinniefalco/DSPFilters/blob/master/shared/DSPFilters/include/DspFilters/Filter.h#L109 ) only supports in-place operation, so this would probably need to be modified aswell. There seems to be a [newer alternative](https://github.com/berndporr/iir1) that might be worth considering.

####  **Use case:**
- I would like to chain together multiple filters, for realtime operation. I think the best way of doing this without causin threading headaches would be to daisy-chain the filters, so that filter N reads from the output buffer of filter N-1 and its output buffer is read by filter N + 1.
- I also think it adds a level of API consistency :)

#### **Describe the solution you'd like:**

I think ideally, for all filters where this makes sense, there should be an implementation that place output either in a new buffer or overwrites the existing data - possibly overwriting if outputData is null. Then there would an overloads that takes just an inputData argument that will overwrite this, and an overload taking an inputdata and outputData argument, that won't overwrite. Both would call the implementation accordingly to perform this.

This means that we have the choice of in-place or using an external buffer, depending on which is more efficient for our use case.

For filters like downsampling or calculating bandpowers, then one can argue that in-place output doesn't make sense, so it wouldn't be necessary to support both modes of operation in these cases.

Andrey1994, 2023-10-01:
There is a genetic rule - if output shape and the meaning of data remains the same after filtering we do calculations in-place. If len is different or if its a completely different data then we return a new array. In realtime apps doing memcpy to return a new array is time and memory consuming, and its the reason to do in-place as much as possible.

Overloads for these methods will be difficult to add because it should be done for all 9 programming languages supported by BrainFlow, and you always can create a copy of array and provide it to filter function instead original data

> Downsampling() overwrites the buffer as it goes along.

Downsampling changes the size of output array, so it cannot be inplace

> get_csp outputs two values, so takes two output buffers as arguments anyway.

This method is not used and I am not sure how to adopt it and even that it still works properly, this requires rethinking

> The rolling filter() [adds samples one-by-one and then overwrites the input buffer.](https://github.com/brainflow-dev/brainflow/blob/5.8.2/src/data_handler/data_handler.cpp#L425)

This done for similarity with other filters, the same for wavelet

> Similarly for the FFT and IFFT

FFT returns completely different types of data, so I do not think that its relevant to methods for filtering

I have some plans for filters redesign mostly to add support for paprallel computations across different channels, I will keep in mind this request during this work

closed 2023-10-01 by Andrey1994

stellarpower, 2023-11-10:
> In realtime apps doing memcpy to return a new array is time and memory consuming

I agree, that's why I'm a bit confused. If we do it in-place, I have to make a second copy of my data. If we output values to a new buffer, I don't need to perform a copy. Is it simply because this is required in the design of the filtering library you're using?

Anyway, if you're going to redesign the filters than I guess that is the priority.

Andrey1994, 2023-11-10:
> If we output values to a new buffer, I don't need to perform a copy. 

Then a memcpy call has to be done inside brainflow all the time, you do not avoid it by moving it into underlying library...
And if all methods are forced to return a new array this memcpy will be unavoidable, while currently if you use smth like `perform_bandpass` you can do it it in-place, and you still can create a new array if needed manually

stellarpower, 2023-11-11:
Well, afraid I still am not following the logic of why a memcpy is strictly needed. If I have a buffer of input values, and I want to process those, personally I would then write the output to a new buffer for the output values, and then I have both to play with. I don't see why I need to perform a copy if I only use the input data once, but nevermind anyway.