brainflow-dev/brainflow #670: Add Support to Polar H10

xrb936, 2023-08-30:
Hi guys! My team is currently working on a project with Muse, EmotiBit, and Polar H10. Currently, Polar H10 heart rate monitor is the only one that is not supported by BrainFlow. I am wondering if it's possible to add it as a new board? It's a general BLE device, which can send heart rate and RR interval information to receiver. Please let me know if I can provide any information you need, thank you!

Andrey1994, 2023-08-30:
Hi!

Its possible but I will need a board and some docs about this device(like how it sends values, how there are encoded, which ble characteristics are used, etc), code samples also will be fine.
If you have contacts with the Polar company you can share them, and it will be good, if you have a device and willing to work on integration by yourself with my help - its even better. Here is a doc to get started https://brainflow.org/2022-11-01-adding-new-boards/ 

I see they have smth on github - https://github.com/polarofficial and it looks promising, I will need to read more about it if I get a device

xrb936, 2023-08-31:
Thank you for the reply. There is a repo on GitHub written by Swift, I think it has everything you need: https://github.com/FleXoft/Polar-H7-HRM

I will also read the document and try to integrate it myself. Is there any device currently supported by BrainFlow also BLE device that I can take a look in the code?

xrb936, 2023-08-31:
I took a look at the code but I am in trouble with understanding how it works. Can you try to modify the code? I would love to help you test it!

Andrey1994, 2023-09-02:
you should use muse board as a reference(not muse bled). High level overview how it works - https://brainflow.readthedocs.io/en/stable/BrainFlowDev.html#navigation 
And there is a PR which does it really good for ble device https://github.com/brainflow-dev/brainflow/pull/668

Andrey1994, 2023-09-02:
You can do the same as in this PR, just ignore part about adding new methods(rotation, etc)