brainflow-dev/brainflow #673: PlaybackBoard significantly slower than StreamingBoard

sjaustirni, 2023-09-09:
**Describe the bug**
I am trying to write a program that can stream files, synthetic and real data alike. I am experiencing a very slow playback stream, which is not the case for other scenarios.

**To Reproduce**
* Stream synthetic data from the OpenBCI interface on `225.1.1.1:6677`
* Run this code with playback set as `True` and `False`

```python
from brainflow.board_shim import BoardShim, BrainFlowInputParams, BoardIds

class Flow:
    def __init__(self, playback=False):
        self.board = None
        BoardShim.enable_dev_board_logger()
        self.playback = playback
        
        self.params = BrainFlowInputParams()
        if self.playback:
            self.params.file = "test.txt"
            self.params.master_board = BoardIds.CYTON_BOARD
        
        else:
            self.params.ip_port = 6677
            self.params.ip_port_aux = 6678
            self.params.ip_address = "225.1.1.1"
            self.params.ip_address_aux = "225.1.1.1"
            self.params.master_board = BoardIds.SYNTHETIC_BOARD
    
    def start(self):
        if self.playback:
            self.board = BoardShim(BoardIds.PLAYBACK_FILE_BOARD, self.params)
        else:
            self.board = BoardShim(BoardIds.STREAMING_BOARD, self.params)
        self.board.prepare_session()
        self.board.start_stream()
    
    def stop(self):
        self.board.stop_stream()
        self.board.release_session()
    
    def _get_data(self):
        return self.board.get_board_data()
    
    def get_user_input(self, data=None):
        if data is None:
            data = self._get_data()
        
        emg_1 = data[1]
        emg_2 = data[2]
        
        if len(emg_1) > 0 or len(emg_2) > 0:
            return emg_1[0], emg_2[0]
        
        return 0, 0

def main():
    flow = Flow(playback=False)
    flow.start()
    
    running = True
    while running:
        left, right = flow.get_user_input()
        if left > 0 or right > 0:
          print(left, right)
    flow.stop()

if __name__ == "__main__":
    main()
```

The test.txt is just an excerpt from the meditation demo from OpenBCI GUI:

```
0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0
0, 0.00, 0.00, 0.00, 0.00, 0.00, 0.00, 0.00, 0.00, 0.014, 0.826, 0.532,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0, 1557936889064, 12:14:49.064
1, 45997.52, 37487.45, -22369.71, -26183.10, -12716.55, -24777.78, -10055.49, -3746.69, 0.014, 0.826, 0.532,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0, 1557936889071, 12:14:49.071
2, 46077.20, 37578.29, -22259.92, -26035.54, -12632.74, -24711.80, -9977.57, -3650.20, 0.014, 0.826, 0.532,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0, 1557936889071, 12:14:49.071
3, 46157.87, 37690.05, -22231.69, -25992.17, -12616.84, -24741.10, -9976.90, -3649.79, 0.014, 0.826, 0.532,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0, 1557936889079, 12:14:49.079
4, 46112.07, 37644.58, -22285.94, -26075.70, -12653.59, -24794.54, -10017.76, -3706.21, 0.014, 0.826, 0.532,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0, 1557936889079, 12:14:49.079
5, 46002.86, 37500.39, -22366.29, -26191.93, -12712.46, -24778.41, -10044.25, -3734.91, 0.014, 0.826, 0.532,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0, 1557936889089, 12:14:49.089
6, 46049.98, 37543.73, -22284.44, -26075.61, -12649.50, -24716.38, -9982.60, -3659.38, 0.014, 0.826, 0.532,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0, 1557936889089, 12:14:49.089
7, 46143.99, 37672.61, -22245.28, -26006.21, -12627.10, -24746.91, -9982.80, -3657.17, 0.014, 0.826, 0.532,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0, 1557936889093, 12:14:49.093
8, 46114.04, 37649.90, -22288.47, -26071.45, -12657.26, -24796.85, -10020.76, -3705.92, 0.020, 0.836, 0.518,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0, 1557936889100, 12:14:49.100
9, 46006.88, 37516.04, -22375.08, -26195.26, -12718.19, -24796.20, -10052.05, -3743.56, 0.020, 0.836, 0.518,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0, 1557936889100, 12:14:49.100
10, 46028.10, 37525.11, -22304.38, -26100.40, -12663.11, -24720.38, -9987.97, -3672.82, 0.020, 0.836, 0.518,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0, 1557936889108, 12:14:49.108
11, 46139.32, 37670.13, -22233.66, -25995.86, -12619.71, -24735.11, -9971.58, -3652.41, 0.020, 0.836, 0.518,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0, 1557936889108, 12:14:49.108
12, 46124.32, 37667.74, -22271.39, -26050.89, -12649.54, -24787.50, -10011.50, -3699.73, 0.020, 0.836, 0.518,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0, 1557936889116, 12:14:49.116
13, 46020.09, 37538.73, -22376.04, -26190.43, -12721.99, -24814.32, -10062.67, -3755.92, 0.020, 0.836, 0.518,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0, 1557936889116, 12:14:49.116
14, 46008.07, 37500.08, -22336.46, -26140.41, -12682.58, -24734.37, -10009.80, -3694.79, 0.020, 0.836, 0.518,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0, 1557936889123, 12:14:49.123
15, 46110.53, 37635.29, -22242.89, -26006.75, -12615.10, -24730.57, -9966.73, -3645.77, 0.020, 0.836, 0.518,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0, 1557936889123, 12:14:49.123
```

**Info (please complete the following information):**
 - Board Id: StreamingBoard (gets Synthetic data from OpenBCI GUI) and PlaybackBoard. No actual hardware used.
 - OS: Ubuntu 22.04.3 LTS x86_64 
 - Programming Language: Python 3.10
 - Architecture: x64
 - `pip list`:
```
 Package    Version
---------- -------
brainflow  5.9.0
nptyping   1.4.4
numpy      1.25.2
pip        22.3.1
pygame     2.5.1
setuptools 65.5.1
typish     1.9.3
wheel      0.38.4
```

**Expected behaviour**
I expect that both streaming board and playback board will send the data in a timely manner, which means around the 250Hz for the streamed synthetic board, and as per timestamps for the playback board.

**Actual behaviour**
While streaming board sends multiple samples each frame (multiple fps) as expected, playback board sends a new sample only every 1–3 seconds (very variable), even though it is apparent from the timestamps that the original data was recorded at multiple hundred Herz.

**Additional context**
It seems that the PlaybackBoard does not have sampling rate available. I would imagine this is something that could be set for the playback scenario? I could not find a way to do that though.

**Debug output** from the streamed scenario:
```
2023-09-09 18:31:38.732] [board_logger] [info] incoming json: {
    "file": "",
    "file_anc": "",
    "file_aux": "",
    "ip_address": "225.1.1.1",
    "ip_address_anc": "",
    "ip_address_aux": "225.1.1.1",
    "ip_port": 6677,
    "ip_port_anc": 0,
    "ip_port_aux": 6678,
    "ip_protocol": 0,
    "mac_address": "",
    "master_board": -1,
    "other_info": "",
    "serial_number": "",
    "serial_port": "",
    "timeout": 0
}
[2023-09-09 18:31:38.732] [board_logger] [trace] Board object created -2
```

**Debug output** from the playback scenario:
```
[2023-09-09 18:32:56.465] [board_logger] [info] incoming json: {
    "file": "test.txt",
    "file_anc": "",
    "file_aux": "",
    "ip_address": "",
    "ip_address_anc": "",
    "ip_address_aux": "",
    "ip_port": 0,
    "ip_port_anc": 0,
    "ip_port_aux": 0,
    "ip_protocol": 0,
    "mac_address": "",
    "master_board": 0,
    "other_info": "",
    "serial_number": "",
    "serial_port": "",
    "timeout": 0
}
[2023-09-09 18:32:56.465] [board_logger] [trace] Board object created -3
```

Is this a bug, or am I just doing something wrong?

Andrey1994, 2023-09-11:
Playback board in brainflow is intended to playback data recorded using brainflow and does not work at all with files recorded in OpenBCI GUI format

Andrey1994, 2023-09-11:
You can record data in brainflow format using OpenBCI GUI file streamer for example, smth like `file://brainflow_data.csv:w`

Andrey1994, 2023-09-11:
it still can be slightly slower than real board or streaming board because its not simple to ensure the exact time gap between two datapoints, but if you use correct file for playback it should work more or less fine

closed 2023-09-11 by Andrey1994