brainflow-dev/brainflow #683: .NET Core app running on MacOS unable to load shared library, using Linux instead of Mac

ttrichar, 2023-10-26:
**Describe the bug**
I have a .NET Core program using the Brainflow nuget package v.5.10.1. I am trying to run this program in VS for Mac, but the prepare_session and enable_dev_board_logger calls seem to be choosing the BoardControllerLibraryLinux as opposed to BoardControllerLibraryMac. I get an error indicating that the libBoardController.so file can't be found.

The error:

`System.DllNotFoundException: Unable to load shared library 'libBoardController.so' or one of its dependencies. In order to help diagnose loading problems, consider setting the DYLD_PRINT_LIBRARIES environment variable: dlopen(liblibBoardController.so, 0x0001): tried: 'liblibBoardController.so' (no such file), '/usr/local/lib/liblibBoardController.so' (no such file), '/usr/lib/liblibBoardController.so' (no such file), '/Users/<user>/Projects/openbiotech-iot-stream-brainflow/openbiotech-iot-stream-brainflow/liblibBoardController.so' (no such file)
   at brainflow.BoardControllerLibraryLinux.prepare_session(Int32 board_id, String input_json)
   at brainflow.BoardControllerLibrary.prepare_session(Int32 board_id, String input_json)
   at brainflow.BoardShim.prepare_session()
   at openbiotech_iot_stream_brainflow.MonitorLoop.MonitorAsync(String[] args) in /Users/<user>/Projects/openbiotech-iot-stream-brainflow/openbiotech-iot-stream-brainflow/QueueHostedService.cs:line 19`

**To Reproduce**
- Clone the following repository using VS for Mac - https://github.com/o-biotech/openbiotech-iot-stream-brainflow
- Add necessary libraries from brainflow compiled_libs: all .dylib and .so files
- Run debugging. Console will ask for a connection string, just type in anything
- Choose a port. At this point, this step doesn't matter, and you can pick whatever
- Error appears when trying to either enable_dev_board_logger or prepare_session

**Info (please complete the following information):**
 - OpenBCI Cyton Board
 - MacOS 12.7.1
 - .NET Core console app
 - Architecture - x64

**Expected behavior**
Mac version of BoardControllerLibrary is used, instead of Linux version

Please let me know if you need any additional info. Thank you!

Andrey1994, 2023-10-26:
Hi, how did you install mono for macos? I see the dockerfile in your repo, is it used?

ttrichar, 2023-10-27:
Hi! The dockerfile is currently not being used, just messing around with a few things. I actually did not install mono, is that a requirement? All I did was use the brainflow nuget package (I also compiled this on my windows machine, and was able to run it successfully)

Andrey1994, 2023-10-29:
I have no mac to test, you can try with mono or with different versions of dotnet
These instruction https://brainflow.readthedocs.io/en/stable/BuildBrainFlow.html#c worked for me on linux in the past

PR #685 (2023-11-08): fix c# binding for linux and macos, add CI tests
closes #683
closed 2023-11-08 by Andrey1994