brainflow-dev/brainflow #687: Cyton Board Sensor Readings have .5 second gaps in reading intervals

ttrichar, 2023-11-15:
**Describe the bug**
I have a Cyton board and a .net core program that is using Brainflow. I've noticed that, when I use "get_board_data", and I look at channel 22 (the timestamp data), there are intermittent gaps of about .5 seconds between two readings. 

**To Reproduce**
This is my main logic on my .net core program:
https://github.com/o-biotech/openbiotech-iot-stream-brainflow/blob/master/openbiotech-iot-stream-brainflow/QueueHostedService.cs

Specifically, I'm looking at this part of the code:
`                board_shim.start_stream();

                int count = 0;

                while (true)
                {
                    System.Threading.Thread.Sleep(5000);

                    Globals.UNPROCESSED_DATA = board_shim.get_board_data();
                    count++;
                    Console.WriteLine();
                    Console.WriteLine($"Data Received, loop number {count}");
                    Console.WriteLine($"Start Time:{Globals.UNPROCESSED_DATA[_timestamp, 0]}");
                    Console.WriteLine($"End Time:{Globals.UNPROCESSED_DATA[_timestamp, Globals.UNPROCESSED_DATA.GetLength(1) - 1]}");

                    await _taskQueue.QueueBackgroundWorkItemAsync(BuildWorkItemAsync);
                }`

The goal of the above code is to poll the Cyton board for 5 seconds, get the data using "get_board_data", and then queue up the result in a background Queue process. Then, the while loop starts over and should pick up at or near the last timestamp in the previous interval. 

However, when I look at the polled data, there are .5 second gaps within the readings themselves, as shown below:

[image]

**Info (please complete the following information):**
 - OpenBCI Cyton Board
 - Windows 11
 - C# .NET Core
 - x64

**Expected behavior**
The timestamps on data should follow a consistent pattern without time gaps

Please let me know if I can provide any additional information. Thank you!

Andrey1994, 2023-11-16:
Hi,first of all https://docs.openbci.com/Troubleshooting/FTDI_Fix_Windows/#:~:text=The%20fix%20is%20easy.,terms%20of%20packet%20time%20intervals.

Also, idk what your code exactly does but it looks like it contributes a lot in the gaps.

To see if there are real brainflow related gaps or not, store data from a single run of get_board_data into a file and check timestamps there.

closed 2023-11-16 by Andrey1994

ttrichar, 2023-11-17:
Hi Andrey,

Thanks for the reply. I initially thought that my looping logic might have had something to do with it too, so I simplified the code to just take a single 5 second sampling of data, and still found that there were .5 second gaps, all within the same polling period.

Edited the code to look like this:

`board_shim.start_stream();

System.Threading.Thread.Sleep(5000);

Globals.UNPROCESSED_DATA = board_shim.get_board_data();`

I then look at the 2d array that is returned in UNPROCESSED_DATA, and found the same gaps within the data

[image]

[image]

[image]

[image]

I still need to look into the latency setting that you mentioned, although I'm not using the OpenBCI GUI, so I'm trying to find how to set the latency directly within my program