brainflow-dev/brainflow #691: Add support for dual OpenBCI cyton+daisy boards (for one brain + one computer)

soulsyrup, 2023-12-18:
**Use case:**
I would like to connect two pairs of cyton+daisy OpenBCI boards to a computer, using BrainFlow

**Describe the solution you'd like:**
Ability for BrainFlow to get the data (of both cyton+daisy boards as one set of signals) and send it out like it does with other hardware

Andrey1994, 2023-12-18:
hi, you can create two instances of BoardShim class and get data from both simultaneously. In terms of multiboard support it should be enough for all cases, if you want you can stack data later on how you want.

Andrey1994, 2023-12-18:
https://github.com/brainflow-dev/brainflow/blob/master/python_package/examples/tests/brainflow_multiboard_get_data.py smth like this

closed 2023-12-18 by Andrey1994