brainflow-dev/brainflow #695: Support Resistance Information from BrainBit

pomofomo, 2024-01-08:
**Use case:**

Maybe this exists but I couldn't find it.

There are a number of resistence channels along with the eeg channels. However, when testing on BrainBit (Python + MacOS), the resistance always returned 0. I wanted to get the actual value so I could use it to show red/yellow/green lights to allow the user to adjust their headset for a good connection before starting the recording.

**Describe the solution you'd like:**

I looked into libneurosdk2 and the documentation shows there are two ways to monitor that cannot happen simultaneously - either get EEG data or get resistence data. I would like some API or documentation (maybe this is some config option?) that would allow to switch between the two. The caller would need to then just check the proper subset (either resistence or eeg).

**Describe alternatives you've considered:**

I guess I could manually add some code for this case that directly calls the embedded libneurosdk2 dll and tries to switch the polling mode. But this seems dangerous to do and likely to mess with the brainbit dll

Andrey1994, 2024-01-08:
Hi, its supported bit should be enabled using special commands and it will replace normal eeg measurements.

Here is an example https://github.com/brainflow-dev/brainflow/blob/master/python_package%2Fexamples%2Ftests%2Fbrainbit_resist.py

closed 2024-01-08 by Andrey1994

pomofomo, 2024-01-08:
Wow! Thank you for that quick response. That is a perfect example