brainflow-dev/brainflow #703: Accelerometer and gyroscope data stream for GforcePro Oymotion band

zubaidah93, 2024-01-24:
**Use case:**
A clear and concise description of what the problem is and why do you need this feature

**Describe the solution you'd like:**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered:**
A clear and concise description of any alternative solutions or features you've considered.

Andrey1994, 2024-01-25:
We dont parse it currently, I will need to check if its available together with the EMG data. If there are no conflicts in sampling rates, etc, we can add it

Andrey1994, 2024-01-28:
[temp.patch](https://github.com/brainflow-dev/brainflow/files/14077636/temp.patch)

I cannot make it work together with emg data, as soon as I enable accel and gyro data on gforceSDKCXX level I stop getting valid emg from the board, temp git diff is attached, so you can patch it a little bit locally depending on your own needs.

I also see that in gforceSDKCXX examples accel and gyro data are off, so maybe there was a reason for it. I would recommend you to ask oymotion guys for more info

closed 2024-01-28 by Andrey1994