brainflow-dev/brainflow #704: ModuleNotFoundError: No module named 'pkg_resources'

Brogoming, 2024-01-26:
This is the issue I am seeing when I try to run my code:
    Traceback (most recent call last):
      File "c:\Users\user\Desktop\EEG_software\PythonGetDatafromaBoard\main.py", line 3, in <module>
        from brainflow.board_shim import BoardShim, BrainFlowInputParams, BoardIds, BrainFlowPresets
      File "C:\Users\user\AppData\Local\Programs\Python\Python312\Lib\site-packages\brainflow\__init__.py", line 1, in <module>
        from brainflow.board_shim import *
      File "C:\Users\user\AppData\Local\Programs\Python\Python312\Lib\site-packages\brainflow\board_shim.py", line 10, in <module>
        import pkg_resources
    ModuleNotFoundError: No module named 'pkg_resources'

The current version of brainflow I have is 5.12.0, I'm trying to see if older versions work. I also have the latest version of setuptools 69.0.3.

Andrey1994, 2024-01-26:
I think it might be smth with your setuptools installation or python, it works fine in CI tests and for me locally also.

Andrey1994, 2024-01-26:
There are relevant discussions on github and stackoverflow with completely different packages and common recommendation there is to fix installation somehow

closed 2024-01-26 by Andrey1994