brainflow-dev/brainflow #708: Failed to install build.py file

laudaraissa, 2024-03-07:
**Describe the bug**
I can't install build.py and it shown "ValueError : Process finished with error code 2". Tried it multiple times but the same error still occurs.

**To Reproduce**
To report a bug you need to create a simple example that uses only BrainFlow API or use one of the tests we already have.  In this example, you need to call `enable_dev_board_logger ()` and share the output with us. Also, make sure the latest version is used. 

**Info (please complete the following information):**
 - Board Id : Raspberry pi Zero W
 - OS running on your PC : Linux
 - Programming Language and its version : python
 - Architecture(x86, x64, ARM, etc) 
 - In case of Python full output of `pip list` command, for other languages please provide BrainFlow version used

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

**Additional context**
Add any other context about the problem here.

Andrey1994, 2024-03-07:
Which gcc version? And what is above the screenshot?

laudaraissa, 2024-03-08:
thank you for responding,

the gcc version on my raspberry pi zero w is **gcc (Raspbian 10.2.1-6++rpi1) 10.2.1 20210110**

and above the screenshot is a command **"python3 build.py"**. I followed all the instructions but the same error always occur, and I can't move on to the next step (for install --U)

Andrey1994, 2024-03-08:
I mean I want to see entire stdout

laudaraissa, 2024-03-09:
my bad, this is the entire script for the stdout.

laudaraissa, 2024-03-09:
I'll attach the file here, in case anyone face the same problem

[Cmake Installation Error.docx](https://github.com/brainflow-dev/brainflow/files/14545074/Cmake.Installation.Error.docx)

Andrey1994, 2024-03-09:
https://github.com/nlohmann/json/issues/1861 as stated here the bedt solution may be to update the gcc

closed 2024-03-09 by Andrey1994