brainflow-dev/brainflow #710: ISSUE when using relaxation session in brainflow version 5.4

armmosikyan66, 2024-03-24:
**Describe the bug**
when i use code sample below I got this error 
Traceback (most recent call last):
  File "/Users/armanmosikyan/PycharmProjects/diplomayin/relaxation.py", line 34, in <module>
    main()
  File "/Users/armanmosikyan/PycharmProjects/diplomayin/relaxation.py", line 26, in main
    relaxation_params = BrainFlowModelParams(BrainFlowMetrics.RELAXATION.value, BrainFlowClassifiers.REGRESSION.value)
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/enum.py", line 429, in __getattr__
    raise AttributeError(name) from None
AttributeError: RELAXATION

**code*
Add any other context about the problem here.
import time
import numpy as np

from brainflow.board_shim import BoardShim, BrainFlowInputParams, LogLevels, BoardIds
from brainflow.data_filter import DataFilter
from brainflow.ml_model import BrainFlowMetrics, BrainFlowClassifiers, BrainFlowModelParams, MLModel

def main():
    params = BrainFlowInputParams()
    params.serial_port = "/dev/cu.usbserial-DM01N8KH"
    board = BoardShim(BoardIds.CYTON_BOARD, params)
    master_board_id = board.get_board_id()
    sampling_rate = BoardShim.get_sampling_rate(master_board_id)
    board.prepare_session()
    board.start_stream()
    BoardShim.log_message(LogLevels.LEVEL_INFO.value, 'start sleeping in the main thread')
    time.sleep(5)  # recommended window size for eeg metric calculation is at least 4 seconds, bigger is better
    data = board.get_board_data()
    board.stop_stream()
    board.release_session()

    eeg_channels = BoardShim.get_eeg_channels(int(master_board_id))
    bands = DataFilter.get_avg_band_powers(data, eeg_channels, sampling_rate, True)
    feature_vector = np.concatenate((bands[0], bands[1]))
    relaxation_params = BrainFlowModelParams(BrainFlowMetrics.RELAXATION.value, BrainFlowClassifiers.REGRESSION.value)
    relaxation = MLModel(relaxation_params)
    relaxation.prepare()
    print('Relaxation: %f' % relaxation.predict(feature_vector))
    relaxation.release()

if __name__ == "__main__":
    main()

Andrey1994, 2024-03-24:
Hi, there was a breaking change in new major release and you should use the latest api ref and code samples and latest brainflow version

closed 2024-03-24 by Andrey1994

armmosikyan66, 2024-03-26:
> Hi, there was a breaking change in new major release and you should use the latest api ref and code samples and latest brainflow version

hi there I get the code example from docs my package version is 5.12.1 and I got the error where I can get the current version code example to calculate the relaxation ?