brainflow-dev/brainflow #713: Publish brainflow on bioconductior

Pandapip1, 2024-04-12:
**Use case:**

Publishing the brainflow R package on [bioconductor](https://bioconductor.org/) will make it much more accessible to install.

**Describe the solution you'd like:**

Brainflow should be published on bioconductor.

**Describe alternatives you've considered:**

If you could get brainflow on CRAN, that would be even better, but that would be a bit of a stretch.

Andrey1994, 2024-04-21:
Hi!

I thought about CRAN but it will still require some manual setup because brainflow R package uses reticulate and calls python code. So proper python setup is required and as far as I know cannot be easily managed by CRAN. So unfortunatelly its not an option.

I will read about bioconductor, have not heard about it before

Andrey1994, 2024-04-22:
seems like the same is the case for bioconductor and for R package in general, so no plans to fix

Pandapip1, 2024-04-23:
Mind updating this issue to be closed as 'not planned' instead of 'completed?'

closed 2024-04-23 by Andrey1994