brainflow-dev/brainflow #715: Muse MU-01 (2014) Not Connecting to BrainFlow

tuckerisapizza, 2024-04-21:
Muse MU-01 (2014) fails to be detected or connect to Brainflow.
(Muse MU-01 is the Muse headband with 2 charging ports, not to be confused with later models such as MU-02)

To Reproduce
Put Muse MU-01 (2014) into pairing mode (5 seconds power button). Run Brainflow code using code muse_2016_board. Brainflow fails to detect Muse over bluetooth.

Info
 - Muse MU-01 (2014)
 - Windows 11
 - Python
 - x64
 - brainflow>=5.12.0

Exact Error
```
[2024-04-20 22:19:41.453] [board_logger] [info] found 1 BLE adapter(s)
[2024-04-20 22:19:51.478] [board_logger] [error] Failed to find Muse Device
Traceback (most recent call last):
  File "C:\Users\Tucker\Documents\BrainFlowsIntoVRChat-main\main.py", line 160, in main
    board, logics, refresh_rate_hz = BoardInit(args)
                                     ^^^^^^^^^^^^^^^
  File "C:\Users\Tucker\Documents\BrainFlowsIntoVRChat-main\main.py", line 125, in BoardInit
    board.prepare_session()
  File "C:\Users\Tucker\AppData\Local\Programs\Python\Python311\Lib\site-packages\brainflow\board_shim.py", line 1190, in prepare_session
    raise BrainFlowError('unable to prepare streaming session', res)
brainflow.exit_codes.BrainFlowError: BOARD_NOT_READY_ERROR:7 unable to prepare streaming session
```

Muse 2014 should be able to use code muse_2016_board (MU-02) as they are essentially the same headband, and connect in the same fashion. No internals have been changed.

Andrey1994, 2024-04-21:
Hi.

I dont have this device to reproduce this error. I can be wrong but I think Muse boards should not be paired with PC, so I would recommend to try wo it

tuckerisapizza, 2024-04-21:
It isn't detected through brainflow even when not paired, it shows on bluetooth, but never connects to brainflow. I've tried on the connecting mode (single press) aswell with no avail.

Andrey1994, 2024-04-22:
seems like muse 2014 has significant differences comparing to muse 2, at the very least different eeg channels, and potentially another connection procedure and maybe different data format(not sure about this).

Muse 2014 is not officially supported and out of stock, so we dont have plans to integrate it into brainflow by ourselves.

But if you manage to fix this issue and willing to send a PR with the fix I will be glad to merge it(it probably should look like a standalone device in terms of brainflow)

closed 2024-04-23 by Andrey1994