brainflow-dev/brainflow #718: Create new release

Pandapip1, 2024-05-01:
**Use case:**

I'm packaging up brainflow for the nix package manager, and am unable to build the latest version (5.12.1) due to https://github.com/brainflow-dev/brainflow/pull/712 not being present.

**Describe the solution you'd like:**

Please publish a new release that successfully builds.

**Describe alternatives you've considered:**

For the time being, I'll point the build at `master`.

Andrey1994, 2024-05-01:
hi, will do, in the meantime you are welcome to create a PR with nix package for brainflow.

Pandapip1, 2024-05-01:
> in the meantime you are welcome to create a PR with nix package for brainflow.

I'm actually upstreaming this to [nixpkgs](https://github.com/NixOS/nixpkgs).

Andrey1994, 2024-05-09:
I want to add #716 to the new release, as soon as its done, will create a new one

Pandapip1, 2024-05-09:
Fair enough!

closed 2024-08-26 by Andrey1994