brainflow-dev/brainflow #720: nptyping.error.InvalidArgumentsError: Unexpected argument of type <class 'type'>.

Pandapip1, 2024-05-07:
## Description

When attempting to import brainflow, I get the following error:

```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "<string>", line 1, in <lambda>
  File "/nix/store/aav6i2sih47qjwz0shpl4mmpp877v16k-python3-3.12.3/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/nix/store/iw1zsfqcacbdrwcs3gd8mas2pa5xv5a8-python3.12-brainflow-unstable-2024-05-01/lib/python3.12/site-packages/brainflow/__init__.py", line 1, in <module>
    from brainflow.board_shim import *
  File "/nix/store/iw1zsfqcacbdrwcs3gd8mas2pa5xv5a8-python3.12-brainflow-unstable-2024-05-01/lib/python3.12/site-packages/brainflow/board_shim.py", line 554, in <module>
    class BoardShim(object):
  File "/nix/store/iw1zsfqcacbdrwcs3gd8mas2pa5xv5a8-python3.12-brainflow-unstable-2024-05-01/lib/python3.12/site-packages/brainflow/board_shim.py", line 1269, in BoardShim
    def get_current_board_data(self, num_samples: int, preset: int = BrainFlowPresets.DEFAULT_PRESET) -> NDArray[Float64]:
                                                                                                         ~~~~~~~^^^^^^^^^
  File "/nix/store/j4sdr7894fvmyvxibygh2ibkcl4l1cvw-python3.12-nptyping-2.5.0/lib/python3.12/site-packages/nptyping/base_meta_classes.py", line 145, in __getitem__
    args = cls._get_item(item)
           ^^^^^^^^^^^^^^^^^^^
  File "/nix/store/j4sdr7894fvmyvxibygh2ibkcl4l1cvw-python3.12-nptyping-2.5.0/lib/python3.12/site-packages/nptyping/ndarray.py", line 73, in _get_item
    cls._check_item(item)
  File "/nix/store/j4sdr7894fvmyvxibygh2ibkcl4l1cvw-python3.12-nptyping-2.5.0/lib/python3.12/site-packages/nptyping/ndarray.py", line 109, in _check_item
    raise InvalidArgumentsError(f"Unexpected argument of type {type(item)}.")
nptyping.error.InvalidArgumentsError: Unexpected argument of type <class 'type'>.
```

## Reproduction

```
$ git clone https://github.com/brainflow-dev/brainflow
$ git checkout 41c304e7545a7465b06201dd4652ef93822edb83 # the commit I'm testing with
$ cd brainflow
$ cmake .
$ make
$ cd python_package
$ python setup.py bdist_egg
$ python
>>> import sys
>>> sys.path.append("./dist/brainflow-0.0.1-py3.12.egg")
>>> import brainflow
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/tmp/brainflow/python_package/brainflow/__init__.py", line 1, in <module>
    from brainflow.board_shim import *
  File "/tmp/brainflow/python_package/brainflow/board_shim.py", line 554, in <module>
    class BoardShim(object):
  File "/tmp/brainflow/python_package/brainflow/board_shim.py", line 1269, in BoardShim
    def get_current_board_data(self, num_samples: int, preset: int = BrainFlowPresets.DEFAULT_PRESET) -> NDArray[Float64]:
                                                                                                         ~~~~~~~^^^^^^^^^
  File "/nix/store/j4sdr7894fvmyvxibygh2ibkcl4l1cvw-python3.12-nptyping-2.5.0/lib/python3.12/site-packages/nptyping/base_meta_classes.py", line 145, in __getitem__
    args = cls._get_item(item)
           ^^^^^^^^^^^^^^^^^^^
  File "/nix/store/j4sdr7894fvmyvxibygh2ibkcl4l1cvw-python3.12-nptyping-2.5.0/lib/python3.12/site-packages/nptyping/ndarray.py", line 73, in _get_item
    cls._check_item(item)
  File "/nix/store/j4sdr7894fvmyvxibygh2ibkcl4l1cvw-python3.12-nptyping-2.5.0/lib/python3.12/site-packages/nptyping/ndarray.py", line 109, in _check_item
    raise InvalidArgumentsError(f"Unexpected argument of type {type(item)}.")
nptyping.error.InvalidArgumentsError: Unexpected argument of type <class 'type'>.
```

## Info

 - Board Id: *N.A.*
 - Operating System: NixOS
 - Programming Language: Python 3.12
 - Architecture: x86_64

```
$ pip list
Package    Version
---------- ------------
nptyping   2.5.0
numpy      1.26.4
pip        24.0
setuptools 69.5.1.post0
```

## Expected behavior

Brainflow should import successfully.

## Screenshots

*N.A.*

## Additional context

*N.A.*

Andrey1994, 2024-05-09:
hi, cannot reproduce, have not tried with nix but normal installation works fine with python 3.11 and 3.12
One of the possible reasons is that versions you have in pip list do not match versions specified in setup.py for brainflow - 
```
install_requires=[
        'numpy',
        'nptyping==1.4.4'
    ],
```

closed 2024-05-09 by Andrey1994

Andrey1994, 2024-05-09:
especially since there is different major version of nptyping

Pandapip1, 2024-05-09:
I'll assume that this is an issue with the version of nptyping.

Andrey1994, 2024-05-09:
in setup.py version is correct, so the issue is in your build env

Pandapip1, 2024-05-09:
I can confirm that the issue was an incorrect version of nptyping. Under `nptyping == 1.4.4`, `brainflow` builds!