brainflow-dev/brainflow #721: Upload brainflow rust bindings to crates.io

Pandapip1, 2024-05-09:
**Use case:**

I'd like to use brainflow in a rust project, and https://crates.io/ is the standard way of distributing Rust packages.

**Describe the solution you'd like:**

Brainflow should be uploaded to https://crates.io/

**Describe alternatives you've considered:**

*N.A.*

Andrey1994, 2024-05-09:
hi, there is one problem I cannot solve with crates. BrainFlow uses C\C++ dyn libs in all bindings and crates does not allow to package binary files properly, there is a strict limitation for max size of the package.
idk how other projects bypass it

Pandapip1, 2024-05-09:
> hi, there is one problem I cannot solve with crates. BrainFlow uses C\C++ dyn libs in all bindings and crates does not allow to package binary files properly, there is a strict limitation for max size of the package.

My understanding is that crates just assume that the user has the libraries installed.

Andrey1994, 2024-05-09:
but in that case users still need to download libraries from somewhere and set LD_LIBRARY_PATH or PATH manually. Then advantage of using crates only for Rust code is debatable

retiutut, 2024-08-30:
It does seem to be the official package distribution for Rust https://github.com/rust-lang/crates.io.

closed 2024-12-28 by Andrey1994

Pandapip1, 2024-12-28:
Was this closed accidentally, or is this actually not planned. If so, would you mind just providing a short explanation for the wontfix, since all your previous concerns have been addressed?

Andrey1994, 2024-12-28:
Wont fix, I dont see much value uploading rust part only wo compiled libs, they still should be copied manually. And afik there is still file size limit.
So until there is no more or less acceptable way to bundle C libs with rust part we will not upload it