brainflow-dev/brainflow #724: Problem getting data from Enophone device on a Mac

basophobic, 2024-06-10:
I've been trying to make the demo program that reads data from the Enophone device and plots it in real time on a Mac for the last couple of days, but with no luck. The program outputs an error message about no data for the last 5 seconds. I have tried an external USB Bluetooth dongle to connect the headphones, which works without problems (pairing and connection are OK).
The same code has been used on my Linux PC without any issues.
Any thoughts or similar behavior from anybody?

Andrey1994, 2024-06-10:
Hi, its the same as here https://github.com/brainflow-dev/brainflow/issues/611

I have no mac to test it, would be good if you contact enophone guys for assistance. I hope they will help with debugging and send a PR to brainflow repo with the fix. I sent them message about it in the past but this issue still exists

basophobic, 2024-06-10:
Thanks, Andrey! I will contact them and let you know! Keep up the excellent job!

Andrey1994, 2024-09-22:
I've tried to use serial port instead sockets on macos, didnt work for me from the box. Since classic bluetooth is quite outdated and been replaced by BLE everywhere, I dont think I will spend more time trying to fix it, closing it as non planned

closed 2024-09-22 by Andrey1994