brainflow-dev/brainflow #726: brainflow.exit_codes.BrainFlowError: BOARD_WRITE_ERROR:4 unable to prepare streaming session

NewBiee777, 2024-06-19:
Hello !
I just started to test Brainflow Python API.
Trying to run the get data from board with my Cyton and its WiFi shield.
See the results.
Am I missing something?
Thanks a lot !
[image]
[image]

Andrey1994, 2024-06-24:
Hi! How do you run it? I dont understand why local ip and default iü from argparse are in different nets..

Andrey1994, 2024-06-24:
And what is the full output from the dev logger?

NewBiee777, 2024-06-26:
> And what is the full output from the dev logger?
This is the complete logger
[image]

Andrey1994, 2024-06-26:
Can you show the full json on top?

NewBiee777, 2024-06-26:
You mean the full list of outputs?
[image]

Andrey1994, 2024-06-26:
how your network is configured? your pc connected to the net from wifi shield, right?

Andrey1994, 2024-06-26:
can you share the output of `ipconfig.exe` executed in the console?

closed 2024-08-26 by Andrey1994