brainflow-dev/brainflow #733: [Node.js] Brainflow not working in electron when electron-app is packaged (node.js)

GordonPmnt, 2024-07-02:
**Describe the bug**

This is more a compatibility issue with the Node.js package and electron.js.

When running brainflow in an electron app, it\s working well in a dev env. But when packaging the app, electron cannot resolve the dll path in `Boardshim` constructor.

```ts
this.libPath = `${__dirname}/../brainflow/lib`;
```

Today, in order to solve this issue, we had to adapt brainflow ourselves:

```ts
    const RESOURCES_PATH = app.isPackaged
      ? path.join(process.resourcesPath, 'app.asar.unpacked', 'node_modules', 'brainflow')
      : path.join(__dirname, '../../brainflow');
    this.libPath = path.join(RESOURCES_PATH, 'brainflow', 'lib');
```

**To Reproduce**

Install brainflow in an electron app. Create a button running an instance of synthetic board. 

Run it locally, it will work.

Then, package the app and run it. Electron shoudl complain because it cannot resolve path to import dll files.

**Info (please complete the following information):**
 - Board Id: any
 - OS running on your PC: any
 - Programming Language and its version: node.js 20
 - Architecture(x86, x64, ARM, etc): any 
 - In case of Python full output of `pip list` command, for other languages please provide BrainFlow version used

**Expected behavior**
Path should be resolved properly in electron packaged context for production.

** suggestion **
Init constructor differently, or add an electron support.

Andrey1994, 2024-08-26:
hi, sorry for late response.

I am not very well familiar with electron and js in general... is there a way to pack libs properly to solve it?

Andrey1994, 2024-08-26:
and will it help if we do smth like this:
```
if (libs in current path exist)
    load libs from it
else
    try to load it wo path specification and then OS should search for it in current dir and in PATH/LD_LIBRARY_PATH env vars
```
?

If so could you send a PR implementing it?

retiutut, 2024-08-30:
I'm familiar with how this plays out using the BrainFlow Java library, and we've written some code in the Java binding to handle this. This may need a new contribution in the NodeJS binding or example code.