brainflow-dev/brainflow #734: WASM build and support for browsers

mpbb, 2024-07-17:
The existence of WebUSB and Web Serial make in browser data collection a viable platform for brainflow.

Can we explore compiling to wasm and perhaps packaging for web?

Andrey1994, 2024-07-18:
I think, I built it for web assembly before. Signal processing part and synthetic board worked fine, but all IO boards were broken. None of the actual boards worked for me from the box and I even not sure that its possible.
No idea how web serial works and not an expert in web...
Would be good if you take the lead on that and help with porting to wasm

Andrey1994, 2024-11-14:
DataFilter part should work, I dont remember how I built it, but more likely smth like this https://peterdn.com/post/2020/12/31/building-c-and-cpp-libraries-for-webassembly/

Its a normal Cmake project, so should be straightforward

Andrey1994, 2024-12-28:
closing as non planned, it should be possible to build it for webassembly but its probably not one of the builds we are going to create ourselves. If there are any issues with it feel free to create new ones

closed 2024-12-28 by Andrey1994