brainflow-dev/brainflow #738: Unable to get data from OpenBCI GUI, `WSAGetLastError is 10060`

IgnaciodelaTorreArias, 2024-08-16:
**Describe the bug**
I'm unable to get data while streaming from the OpenBCI GUI.

 - CYTON_BOARD, STREAMING_BOARD
 - Windows 11 pro, Version	10.0.26120 Build 26120
 - Python 3.12.4
 - Architecture: x64
 -  pip list: 

Package | Version |
--- | --- |
brainflow | 5.13.1 |
numpy | 2.0.1 |
pip | 24.2 |
setuptools | 72.2.0 |

**Expected behavior**
Get more than 0 on data count.

**Screenshots**
[image]

Logs from OpenBCI GUI indicate that the streaming is working:

```

OpenBCI_GUI: BrainFlow Streamer Location: streaming_board://225.1.1.1:6677
Brainflow start streaming
[DEFAULT]: Data stream started.

```

[image]

output:

```

[2024-08-15 18:45:47.939] [board_logger] [info] incoming json: {
    "file": "",
    "file_anc": "",
    "file_aux": "",
    "ip_address": "225.1.1.1",
    "ip_address_anc": "",
    "ip_address_aux": "",
    "ip_port": 6677,
    "ip_port_anc": 0,
    "ip_port_aux": 0,
    "ip_protocol": 0,
    "mac_address": "",
    "master_board": 0,
    "other_info": "",
    "serial_number": "",
    "serial_port": "",
    "timeout": 0
}
[2024-08-15 18:45:47.941] [board_logger] [trace] Board object created -2
board.is_prepared() = True board.get_board_data_count() = np.int32(0)
[2024-08-15 18:45:52.950] [board_logger] [trace] unable to read 576 bytes, read -1
[2024-08-15 18:45:52.951] [board_logger] [error] WSAGetLastError is 10060
[2024-08-15 18:45:52.951] [board_logger] [error] socket operation error code: -1

```

Deactivated windows firewalls just in case (didn't work):
[image]

Andrey1994, 2024-08-16:
hi, it seems like you configured everything correctly.

Could you try with two python scripts and in first one add `board.add_streamer("streaming_board://225.1.1.1:6677")` and use it instead openbci gui(set there board id to cyton and serial_port to COM3 probably). It will help to understand where the problem is.

Also, @retiutut could you take a look?

Andrey1994, 2024-08-16:
And is it on the same PC? Or two PCs in the same network? Could you tell about network, is it smth in the university?

Andrey1994, 2024-08-16:
Tried with two python scripts(slightly modified version of brainflow_get_data with two lines for board.add_streamer for two presets) and with streaming_test, works fine for me. 

[image]

But when I run the streamer via gui 5.2.2 it does not, need help from Richard

IgnaciodelaTorreArias, 2024-08-17:
Hi so apparently this is a bug on OpenBCI GUI, issue: [1190](https://github.com/OpenBCI/OpenBCI_GUI/issues/1190) using [v6.0.0-beta.1](https://github.com/OpenBCI/OpenBCI_GUI/tree/v6.0.0-beta.1) solves the problem, sorry for bringing it here, by the logs i thought the problem was on my side.

closed 2024-08-17 by IgnaciodelaTorreArias